russhughes/st7789_mpy #99: How can I create vector fonts?

FabriceAuz, 2022-11-01:
It's all in the title: I'd like to create my own vector fonts, from font files on my computer. How to do that?
Thanks.

russhughes, 2022-11-10:
I have not used this but take a look at https://github.com/LingDong-/ttf2hershey

russhughes, 2022-11-10:
Here is more information on the scripts I used to convert to hershey fonts to python modules:
https://github.com/russhughes/hershey-scripts

russhughes, 2022-11-10:
This code looks messy and outdated; I'll see if I can post something better.

FabriceAuz, 2022-11-11:
Thanks a lot @russhughes I'll try that soon.
If I understand well:

- First use LingDong's script to convert a ttf font to Hershey (as an intermediate representation)
- Then use map2py to obtain the vector font file usable on the ST7789 display

If I obtain good results, do you want me to send you the new vector fonts, so that you can add them on your github ?

FabriceAuz, 2022-11-11:
Actually, I expected it to be much easier, but I can't get the Hershey conversion script to work correctly. I get an erro message saying:

> Unimplemented: Encoding other than (PlatformID = 0, Format = 4) is not yet supported

And indeed, when I print intern variables, I get:
```
0
format= 0
PlatformID= 1
1
format= 4
PlatformID= 3
```

Are these `format `and `PlatformID `dependant on the ttf file? Should I try until I find a ttf file with the correct values?

russhughes, 2022-11-11:
I’m working on some fixes and updates to the conversion programs, I have been able to get the example Ubuntu.ttf to convert and it looks good, another is out of order.  I hope to have an update today.

FabriceAuz, 2022-11-11:
Great !

russhughes, 2022-11-11:
Working on the scripts, here is the converted Ubuntu font. 

[ubuntu.py.zip](https://github.com/russhughes/st7789_mpy/files/9992634/ubuntu.py.zip)

[image]

FabriceAuz, 2022-11-14:
Hi Russ
I have seen your new code `hershey_to_py.py` but it seems there are a few mistakes in it:

- lines 106 and 165: does `:=` mean `!=` ?
- in functions `parse_line` and `hershey_load`, the variable `match`is undefined.

Can you take a look?

russhughes, 2022-11-14:
That's an [assignment expression](https://peps.python.org/pep-0572/) available since Python 3.8 and more recent versions of MicroPython.  I've replaced them with the equivalent code for older pythons.

FabriceAuz, 2022-11-15:
Thanks @russhughes, it works fine. I've been able to create new font files. I'll try them soon on the ST7789, and keep you informed.

lesept777, 2022-11-19:
Hi @russhughes 
I have transformed the fonts from my computer into vector files usable on this display. All the files are here: https://github.com/lesept777/Vector-fonts-for-ST7789-display

Fell free to take any file you want and add them in your gitHub. Thanks a lot for your help. It was great !

One last question: do you think you could come up with a method to display the text using these vector fonts, and fill the letters in a different colour? There would be one colour for the inside, and one for the outline of the letters.

russhughes, 2022-11-23:
I was thinking of filling in the letters by converting the font into polygons instead of vectors. They could be filled quickly. I'll play around with this idea.

lesept777, 2022-11-24:
Thanks, that would be great!

closed 2023-04-27 by russhughes