russhughes/st7789_mpy #101: I compile esp32c3_usb does not work properly

whossun, 2022-11-15:
I tried to compile the firmware for esp32c3_usb, the firmware compiled successfully, but after burning my development board does not work properly, can you compile a firmware for esp32c3_usb?

russhughes, 2022-11-16:
I've compiled and uploaded the firmware. I don't have a device to test the firmware on.

whossun, 2022-11-16:
Thank you very much for helping me compile the firmware for esp32c3_usb, now it works fine on my development board, but I still can't compile the firmware myself to work properly. 
My build environment in Ubuntu 20.04.5, IDF v4.4, python v3.8.10, the latest micropython.
 I compiled as follows:
```
cd esp-idf
git checkout v4.4
git submodule update --init --recursive
./install.sh
source export.sh

cd micropython
git submodule update --init
make -C  mpy-cross

cd micropython/ports/esp32
make BOARD=GENERIC_C3_USB USER_C_MODULES=../../../../st7789_mpy/st7789/micropython.cmake
```
Is something wrong?

russhughes, 2022-11-16:
Try: 
make BOARD=GENERIC_C3_USB USER_C_MODULES=../../../../st7789_mpy/st7789/micropython.cmake submodules all

whossun, 2022-11-17:
I compile micropython without any changes and it works fine, trying to add the st7789 module compiles successfully but does not work, I'm crushed, I tried on different computers with the same result.

russhughes, 2022-11-23:
During the compile, you should see a line saying, `Found User C Module(s): usermod_st7789`. If this line is not printed, the build process is not finding the USER_C_MODULES directory.  Try setting USER_C_MODULES to the full pathname of the micropython.cmake file in the st7789 folder.

whossun, 2022-12-04:
Today I switched idf to v4.4.3 and compiled it, and after compiling it and brushing it into my esp32c3_usb development board, it actually works! The previous compilation under idf v4.4 always failed to support.

closed 2022-12-16 by whossun