russhughes/st7789_mpy #107: LoadProhibited Panic on tft.init() on tdisplay_esp32

LexxM3, 2022-12-09:
- Hardware: `Lilygo TTGO T-Display with ESP32 and 135x240 1.14 inch LCD`
- Firmware image: `MicroPython v1.19.1-630-gd0687e90a-dirty on 2022-11-15; TTGO T-Display with ESP32`
- Setup: correct `examples/configs/tdisplay_esp32/tft_config.py` used
- Test: `examples/hello.py`
- Problem: crash immediately upon execution of `tft.init()`

- Simplest manual reproduction:
```
>>> import st7789
>>> import tft_config
>>> import vga1_bold_16x32 as font
>>> tft = tft_config.config(0)
>>> tft.init()
```

- Details of crash:
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008e1bc  PS      : 0x00060d30  A0      : 0x801011f6  A1      : 0x3ffd0500  
A2      : 0x00000000  A3      : 0xffffffff  A4      : 0x0000000c  A5      : 0x00000000  
A6      : 0x00000008  A7      : 0x00000008  A8      : 0x00000020  A9      : 0x3ff44000  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000005  A13     : 0x3ffd061c  
A14     : 0x0000002c  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000054  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace:0x4008e1b9:0x3ffd05000x401011f3:0x3ffd0530 0x400d2a97:0x3ffd05c0 0x400d2d95:0x3ffd0600 0x400d3436:0x3ffd0750 0x400d548a:0x3ffd07a0 0x400e6921:0x3ffd07f0 0x400ece62:0x3ffd0810 0x400ecf8e:0x3ffd0830 0x400854b9:0x3ffd0850 0x400e6a80:0x3ffd08f0 0x400ece62:0x3ffd0950 0x400ece8a:0x3ffd0970 0x401192c7:0x3ffd0990 0x401194c5:0x3ffd0a20 0x400fd692:0x3ffd0a70 

ELF file SHA256: 8a012495a18b28c1

Rebooting...
```

closed 2022-12-09 by russhughes

LexxM3, 2022-12-09:
I will try lower baud rates, but I AM using your precompiled firmware binaries, so the correlation asserted here is a bit confusing. 

Also, if the default rate doesn't work in such an obscure way, even just some times, it should probably not be the default.

russhughes, 2022-12-09:
The newest precompiled firmware has the fix that was missing.

LexxM3, 2022-12-10:
Ah, apologies, I missed that firmware.bin also got updated (non-source files don't visually stand out in commit histories).  I can confirm that `MicroPython v1.19.1-738-gb042fd512-dirty on 2022-12-09; TTGO T-Display with ESP32` does, in fact, fix this issue and `examples/hello.py` now works as expected!

Thanks for this rapid fix!

And thanks for creating and maintaining this repository in the first place!!!  It is the only coherent source of ready support for ESP32 display boards in Micropython as of Dec 2022 (all the rest, including from Lilygo, are long abandoned and/or so out of date as to be unusable).  It would be more than a great idea to have this be incorporated in official Micropython releases in some maintained form -- is this something that your users can help to "encourage" somehow?

russhughes, 2022-12-12:
Preconfigured tft_config.py modules for several devices can be found in the examples/configs/ folders.  If your device is not listed, you will have to modify one to match your device.

numeru55, 2022-12-12:
Works tft_config.py well! Thanks!

commit 50f91804 (2022-12-09): Fix #107, SPI Baudrates above 26.7MHz