russhughes/st7789_mpy #110: Compil on micropython 1.18

l-atome, 2022-12-21:
Hello, i try to construct a firmware for the esp32 with the ST7789 module and the usqlite module ...
It seems that the usqlite module is only compilable with 1.18 micropython ...

I try to aggregate st7789 but i'm blocking on some step ...

First, i try to compil it for ports/esp32 without success ... So i begin to construct it for ports/unix ...

1) my first error was :
  ---
In file included from ../../py/objmodule.c:36:
build-standard/genhdr/moduledefs.h:8:2: error: #else after #else
 ---

and when i opened the "moduledefs.h", i saw that 
 ---
#if (mp_module_st7789, MODULE_ST7789_ENABLE)
    extern const struct _mp_obj_module_t mp_module_st7789);
#else
MP_REGISTER_MODULE(MP_QSTR_st7789;
    #define MODULE_DEF_MP_QSTR_ST7789 { MP_ROM_QSTR(MP_QSTR_st7789), MP_ROM_PTR(&mp_module_st7789);
#else
MP_REGISTER_MODULE(MP_QSTR_st7789) },
#else
    #define MODULE_DEF_MP_QSTR_ST7789
#endif
 ---

I success this step by manually change this part of this file by :

 ---
extern const struct _mp_obj_module_t mp_module_st7789;
#define MODULE_DEF_MP_QSTR_ST7789 { MP_ROM_QSTR(MP_QSTR_st7789), MP_ROM_PTR(&mp_module_st7789) },
MP_REGISTER_MODULE(MP_QSTR_st7789, mp_module_st7789, 1)
 ---

2) Now i stop on a new error :
 ---
/home/user/Desktop/aEffacer/modules/st7789/st7789.c: In function 
‘st7789_ST7789_blit_buffer’:
../../py/misc.h:43:24: error: comparison of integer expressions of different signedness: ‘size_t’ {aka ‘long unsigned int’} and ‘mp_int_t’ {aka ‘long int’} [-Werror=sign-compare]
   43 | #define MIN(x, y) ((x) < (y) ? (x) : (y))
 ---

Do you think, i can success to compile the st7789 driver for micropython 1.18 ...
Thanks 
Thomas

russhughes, 2022-12-21:
I was able to compile the driver for micropython 1.18 using esp-idf v4.4.3 by replacing the last five lines in st7789.c with:

```
#if !defined(MICROPY_VERSION) || MICROPY_VERSION <= 70144
MP_REGISTER_MODULE(MP_QSTR_st7789, mp_module_st7789, MODULE_ST7789_ENABLE);
#else
MP_REGISTER_MODULE(MP_QSTR_st7789, mp_module_st7789);
#endif
```

I did not see the sign compare error. The error may be related to the version of the esp-idf you are using.

l-atome, 2022-12-30:
Hello ... Really thanks for this fix ... The compilation works with my config :
* python3.8
* gcc8
* micropython 1.18
* esp-idf 4.2

But when i upload the firmware to my board and execute my code, i have this errors :
`ImportError: no module named 'st7789'`

I compil the firmware with the line command : 
`make USER_C_MODULES=~/modules/micropython.cmake FROZEN_MANIFEST="" FROZEN_MPY_DIR=$UPYDIR/modules `

and the micropython.cmake content is : 
```
include(${CMAKE_CURRENT_LIST_DIR}/usqlite/micropython.cmake)
include(${CMAKE_CURRENT_LIST_DIR}/st7789/micropython.cmake)
```
and i have access to the "usqlite" module ...

I tried also to compil with only the "st7789" modules but i reach the same issue

You will find the log of the compil in this file ....
[make.log](https://github.com/russhughes/st7789_mpy/files/10324727/make.log)

Thanks for your help
Thomas

l-atome, 2023-01-21:
Hello, i arrived to compil the firmware with your driver ...

I think that my problem is to download MicroPython with this command line :
`git clone -b v1.18 --recurse-submodules https://github.com/micropython/micropython.git`
That's append also with other module ??? Do you have an idea why ??

I don't understand when the usage of "-b" because i used it for ESP-IDF and never product a bug ...

I product the project "arkanoid" for my student ... I attach it to this message if you want to use it in your examples, I can "clean" a little the code ...
I work to avoid the blink when we erase the old rectangle when moving ... I create the class "Rectangle" to erase only what we need to avoir blink
You can see the result here :
https://chinginfo.fr/999-video2.mp4

Thanks for all your work
Thomas

[arkanoid.txt](https://github.com/russhughes/st7789_mpy/files/10472242/arkanoid.txt)

closed 2023-01-21 by l-atome