russhughes/st7789_mpy #114: Aliexpress 1.3" Without CS Pin

spooknik, 2023-01-27:
Hello

I have bought a [240x240 1.3" screen from Aliexpress](https://www.aliexpress.com/item/1005003681311991.html) and am having trouble getting it to work. 

It doesn't have a CS pin but is the same component as the Waveshare 1.3 just on a different PCB, so my config looks like:

```
"""Aliexpress 1.3" Display"""

from machine import Pin, SPI
import st7789

TFA = 0	    # top free area when scrolling
BFA = 80	# bottom free area when scrolling

def config(rotation=0, buffer_size=0, options=0):
    return st7789.ST7789(
        SPI(1, baudrate=62500000, sck=Pin(10), mosi=Pin(11)),
        240,
        240,
        reset=Pin(12, Pin.OUT),
        cs=Pin(9, Pin.OUT),
        dc=Pin(8, Pin.OUT),
        backlight=Pin(13, Pin.OUT),
        rotation=rotation,
        options=options,
        buffer_size=buffer_size)
```

There are no errors on running, just nothing shows up on the screen. I double checked all the pins are correct.

ExperiMentor, 2023-01-27:
There's a CS pin shown on the photo on the link you sent?
Without CS (chip select) enabled, the display will stay blank.

spooknik, 2023-01-27:
Confusingly, there are two screens on this product listing. The 0.96" has a CS pin, but the 1.3" does not. 

[image]

ExperiMentor, 2023-01-27:
I'm inclined to agree with your 'Edit' . If you Google 'spi display has no cs' you'll find lots of similar Q's and some suggestions. One that seemed relevant is that you may need to pull RES high (10k resistor or connect it to a spare GPIO?)

spooknik, 2023-01-27:
Yes, I have seen here someone [modified](https://www.instructables.com/Adding-CS-Pin-to-13-LCD/) their screen to get the CS pin. I guess that's always a possibility. 

I found out that [this library](https://github.com/russhughes/st7789py_mpy) works fine with the screen.

[image]

 I'll try to adapt the code below into the config_txt and see if that can give any results.
My code looks like: 

```
"""
noto_fonts Writes the names of three Noto fonts centered on the display
    using the font. The fonts were converted from True Type fonts using
    the font2bitmap utility.
"""

from machine import SoftSPI, Pin
import st7789py as st7789
import time
from fonts import NotoSansMono_32 as noto_mono

def main():

    def center(font, string, row, color=st7789.WHITE):
        screen = tft.width                        # get screen width
        width = tft.write_width(font, string)     # get the width of the string
        if width and width < screen:              # if the string < display
            col = tft.width // 2 - width // 2     # find the column to center
        else:                                     # otherwise
            col = 0                               # left justify

        tft.write(font, string, col, row, color)  # and write the string

    try:
        spi = SoftSPI(
            baudrate=20000000,
            polarity=1,
            phase=0,
            sck=Pin(10),
            mosi=Pin(11),
            miso=Pin(13))

        tft = st7789.ST7789(
            spi,
            240,
            240,
            reset=Pin(12, Pin.OUT),
            cs=Pin(5, Pin.OUT),
            dc=Pin(8, Pin.OUT),
            backlight=Pin(4, Pin.OUT),
            rotation=1)

        # enable display and clear screen
        tft.fill(st7789.BLACK)

        row = 16

        # center the name of the first font, using the font
        #center(noto_sans, "NotoSans", row, st7789.RED)
        #row += noto_sans.HEIGHT

        # center the name of the second font, using the font
        #center(noto_serif, "NotoSerif", row, st7789.GREEN)
        #row += noto_serif.HEIGHT

        # center the name of the third font, using the font
        
       
        while True:
            center(noto_mono, "Hello", row, st7789.RED)
            
    finally:
        # shutdown spi
        if 'spi' in locals():
            spi.deinit()

main()

```

spooknik, 2023-01-27:
Not totally sure what did it, but this config is working with the hello.py example now. I probably just needed to look at it from a different angle.  

```
from machine import Pin, SPI
import st7789

TFA = 0	    # top free area when scrolling
BFA = 80	# bottom free area when scrolling

def config(rotation=0, buffer_size=0, options=0):
    return st7789.ST7789(
        SPI(1, baudrate=30000000, polarity=1, phase=0, sck=Pin(10), mosi=Pin(11)),
        240,
        240,
        reset=Pin(12, Pin.OUT),
        cs=Pin(9, Pin.OUT),
        dc=Pin(8, Pin.OUT),
        backlight=Pin(4, Pin.OUT),
        rotation=rotation,
        options=options,
        buffer_size=buffer_size)
```

I'll close this issue now. Thanks for chiming in with your input ExperiMentor, much appreciated.

closed 2023-01-27 by spooknik