russhughes/st7789_mpy #117: Fast ST7789 only works after initing with ST7789py

negagen, 2023-02-20:
Hi, I build have a board that I made using a ESP32-WROOM, and have a screen that uses ST7789, with ST7789py the screen works well but when I use the fast module it remains blank, after testing a little I found that if I first create a ST7789py example run it and then use the Fast module it works, I don't know why this is happening, I made the binaries using docker here is my Dockerfile.

`
FROM ubuntu:latest

RUN apt update && apt-get -y upgrade && apt -y install build-essential libffi-dev git pkg-config cmake virtualenv python3-pip python3-virtualenv

WORKDIR /micropython

RUN git clone -b v4.4 --recursive https://github.com/espressif/esp-idf.git

RUN cd esp-idf/ && git checkout v4.4 && git pull origin v4.4

SHELL [ "/bin/bash", "-c"]

RUN cd esp-idf/ && ./install.sh

RUN apt-get -y install usbutils

RUN source /micropython/esp-idf/export.sh

RUN git clone https://github.com/micropython/micropython.git

RUN git clone https://github.com/russhughes/st7789py_mpy.git

RUN git clone https://github.com/russhughes/st7789_mpy.git

RUN git clone https://github.com/robert-hh/BMP085_BMP180.git

RUN cd micropython/ && git submodule update --init && cd mpy-cross/ && make

RUN cp st7789_mpy/fonts/bitmap/*.py micropython/ports/esp32/modules
RUN cp st7789_mpy/fonts/truetype/*.py micropython/ports/esp32/modules
RUN cp st7789_mpy/fonts/vector/*.py micropython/ports/esp32/modules

RUN cp st7789py_mpy/lib/st7789py.py micropython/ports/esp32/modules

RUN cp BMP085_BMP180/bmp085.py micropython/ports/esp32/modules

CMD source /micropython/esp-idf/export.sh && \
    cd micropython/ports/esp32 && \
    echo "Compiling firmware" && \
    make USER_C_MODULES=/micropython/st7789_mpy/st7789/micropython.cmake && \
    echo "Copying file micropython.bin to /output" && \
    cp /micropython/micropython/ports/esp32/build-GENERIC/micropython.bin /output/micropython.bin && \
    cp /micropython/micropython/ports/esp32/build-GENERIC/bootloader/bootloader.bin /output/bootloader.bin && \
    cp /micropython/micropython/ports/esp32/build-GENERIC/partition_table/partition-table.bin /output/partition-table.bin && \
    echo "Done"
`

After erasing the flash I use this to upload 

`
esptool.py -p COM3 -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 ./output/bootloader.bin 0x8000 ./output/partition-table.bin 0x10000 ./output/micropython.bin
`

It works perfect with the odd behaviour above, I appreciate any help I can get.

negagen, 2023-02-20:
I don't even need to run an example only adding this to the tft_config.py works.
```
from machine import Pin, SPI, freq
import st7789
import st7789py
import gc

def config(rotation=0, buffer_size=0, options=0):
    spi = SPI(2,
        baudrate=16000000, 
        polarity=1, 
        phase=1)
    
    spi.init(baudrate=16000000, polarity=1, phase=1)
    gc.enable()

    reset = Pin(0, Pin.OUT)
    dc = Pin(2, Pin.OUT)

    screen = st7789py.ST7789(
        spi,
        240,
        240,
        reset=reset,
        dc=dc,
        rotation=rotation)

    screen = st7789.ST7789(
        spi,
        240,
        240,
        reset=reset,
        dc=dc,
        rotation=rotation,
        options=options,
        buffer_size=buffer_size)
    
    return screen
```

russhughes, 2023-02-21:
Are you calling the init() method after the tft_config?

negagen, 2023-02-23:
> Are you calling the init() method after the tft_config?

Yes, I'm running the example hello.py

jd3096-mpy, 2023-06-12:
I have the same question,and I found that reset pin can't be set to GPIO0 ,if you change your resetpin ,you will found it works.

russhughes, 2023-06-12:
I see the issue st7789.c line 71-81: 
```
#define RESET_LOW()                         \
    {                                       \
        if (self->reset)                    \
        mp_hal_pin_write(self->reset, 0);   \
    }

#define RESET_HIGH()                        \
    {                                       \
        if (self->reset)                    \
        mp_hal_pin_write(self->reset, 1);   \
    }
```
When reset is 0, the if(self->reset) is always false. I need to make changes to make pin 0 usable.

jd3096-mpy, 2023-06-13:
Yes,now you can set reset to other GPIO,or just skip this parameter.

ccoupe, 2023-09-19:
I seem to have the same problem of Fast st7789 not working. I'm using a esp32-s3 and a waveshare 2".  Like @negagen, If I create with st7789py first and then with st7789 (fast) it works (although it is not a lot faster in my test)

```py
from machine import Pin, SoftSPI
import st7789py as st7789py
import st7789 as st7789
import deja48 as font

BACKLIGHT_PIN = 39
RESET_PIN = 40
DC_PIN = 41
CS_PIN = 42
CLK_PIN = 2
DIN_PIN = 1 # lower left corner
MISO_PIN = 39   # not used for a display - conflicts with backlight - its weird.
print('create spi')
spi = SoftSPI(baudrate=40000000, polarity=1,
                sck=Pin(CLK_PIN), 
                mosi=Pin(DIN_PIN),
                miso=Pin(MISO_PIN))
print('create tft 1')
tft = st7789py.ST7789(spi, 320, 240,
        cs=Pin(CS_PIN, Pin.OUT), 
        reset=Pin(RESET_PIN, Pin.OUT),
        dc=Pin(DC_PIN, Pin.OUT),
        backlight=Pin(BACKLIGHT_PIN, Pin.OUT),
        rotation=1)

print('create tft 2')
tft = st7789.ST7789(spi, 320, 240,
        cs=Pin(CS_PIN, Pin.OUT), 
        reset=Pin(RESET_PIN, Pin.OUT),
        dc=Pin(DC_PIN, Pin.OUT),
        backlight=Pin(BACKLIGHT_PIN, Pin.OUT),
        rotation=1)

print('displaying')
tft.write(font, "Hello Worldz~",10, 0)
tft.write(font, "Hello Worldz~",10, 50, st7789.RED, st7789.BLACK)
tft.write(font, "Hello Worldz~",10, 100, st7789.BLUE, st7789.BLACK)
tft.write(font, "Hello Worldz~",10, 150, st7789.GREEN, st7789.BLACK)
print('all done')
```
Note:  deja48 is a Truetype font (DejaVuSans)

russhughes, 2023-09-20:
@ccoupe Add a `tft.init()` to initialize the display driver. The pure python driver does this automatically, the C driver does not. 

```
from machine import Pin, SoftSPI
import st7789 as st7789
import deja48 as font

BACKLIGHT_PIN = 39
RESET_PIN = 40
DC_PIN = 41
CS_PIN = 42
CLK_PIN = 2
DIN_PIN = 1 # lower left corner
MISO_PIN = 39   # not used for a display - conflicts with backlight - its weird.

print('create spi')
spi = SoftSPI(baudrate=40000000, polarity=1,
                sck=Pin(CLK_PIN), 
                mosi=Pin(DIN_PIN),
                miso=Pin(MISO_PIN))

print('create tft')
tft = st7789.ST7789(spi, 320, 240,
        cs=Pin(CS_PIN, Pin.OUT), 
        reset=Pin(RESET_PIN, Pin.OUT),
        dc=Pin(DC_PIN, Pin.OUT),
        backlight=Pin(BACKLIGHT_PIN, Pin.OUT),
        rotation=1)

print('initialize display driver')
tft.init()

print('displaying')
tft.write(font, "Hello Worldz~",10, 0)
tft.write(font, "Hello Worldz~",10, 50, st7789.RED, st7789.BLACK)
tft.write(font, "Hello Worldz~",10, 100, st7789.BLUE, st7789.BLACK)
tft.write(font, "Hello Worldz~",10, 150, st7789.GREEN, st7789.BLACK)
print('all done')
```

ccoupe, 2023-09-20:
Thanks @russhughes - that was it.  Is there a problem with rotation? I can set it to `1` but it doesn't do what the slow driver does.

I also found the name change from `write_width` to `write_len`.

ccoupe, 2023-09-20:
Never mind about the rotation question. It seems w and h might be different between slow and fast creation

Fails: `tft = st7789.ST7789(spi, 320, 240, `

Works: `tft = st7789.ST7789(spi, 240, 320,`

russhughes, 2023-09-20:
@ccoupe There are slight differences between the two drivers. If you are using an ESP32-S3 with PSRAM take a look at https://github.com/russhughes/s3lcd. It's a newer driver with more features as well as faster DMA based drawing.

closed 2023-09-22 by russhughes