russhughes/st7789_mpy #121: aliexpress 172x320 display has border artefacts

stoneblock, 2023-03-19:
the firmware works on a raspberry pico with a 172x320 pixel display from aliexpress
but at the border one "wrong" row of pixel on each side (difference 170 to 172) is showing

at the adafruit sources for circuit python the only difference between 170x320 and 172x320 is 
BORDER_WIDTH = 28 vs. BORDER_WIDTH = 20
[st7789_172x320_1.47_simpletest.py vs. st7789_170x320_1.9_simpletest.py]
(https://github.com/adafruit/Adafruit_CircuitPython_ST7789/blob/main/examples/st7789_172x320_1.47_simpletest.py) 
(https://github.com/adafruit/Adafruit_CircuitPython_ST7789/blob/main/examples/st7789_170x320_1.9_simpletest.py)

sorry I can't make it by myself maybe it's a small change.

stoneblock, 2023-03-19:
"""Aliexpress Pico LCD 1.47 inch display"""
#wrong display width is set to 170 should be 172 but then the display is worst
#the helping information about offset don't help me without an working example ...
# {  (inversion_mode(False) - color_order = st7789.BGR - for rotation 0 use offset(2, 1) ...  } 

from machine import Pin, SPI
import st7789

TFA = 0	 # top free area when scrolling
BFA = 0	 # bottom free area when scrolling

def config(rotation=0, buffer_size=0, options=0):
    return st7789.ST7789(
        SPI(1, baudrate=62500000, sck=Pin(10), mosi=Pin(11)),
        170,
        320,
        reset=Pin(12, Pin.OUT),
        cs=Pin(14, Pin.OUT),
        dc=Pin(13, Pin.OUT),
        backlight=Pin(8, Pin.OUT),
        rotation=rotation,
        options=options,
        buffer_size=buffer_size)

russhughes, 2023-03-19:
I don't have this display to test but using a custom rotations list should help. 

```
"""
Adafruit 1.47" 320x172 Round Rectangle Color IPS TFT Display - Product ID: 5393
"""

from machine import Pin, SPI
import st7789

TFA = 0
BFA = 0

custom_rotations = [
    (0x00, 172, 320, 34, 0),
    (0x60, 320, 172, 0, 34),
    (0xc0, 172, 320, 34, 0),
    (0xa0, 320, 172, 0, 34),
]

def config(rotation=0, buffer_size=0, options=0):
    return st7789.ST7789(
        SPI(2, baudrate=40000000, sck=Pin(10), mosi=Pin(11), miso=None),
        172,
        320,
        reset=Pin(12, Pin.OUT),
        cs=Pin(14, Pin.OUT),
        dc=Pin(13, Pin.OUT),
        backlight=Pin(8, Pin.OUT),
        rotations=custom_rotations,
        rotation=rotation,
        options=options,
        buffer_size= buffer_size)
```

closed 2023-04-10 by russhughes