russhughes/st7789_mpy #125: font2bitmap.py creates fonts that do not load?

jouellnyc, 2023-04-10:
Hello. 

Following [this doc](https://penfold.owt.com/st7789py/fonts.html#true-type-font-conversion), I tried to convert a few different TTFs.

[this one](https://www.fontrepo.com/font/13715/scoreboard.ttf) for example.

I can write to the screen using the provided  vga2_8x16 font just fine:

```
>>> import fonts.vga2_8x16 as font
>>> d.text(font, 'hi', 10, 10)

```

If I try using the converted font I get:

>>> 
import fonts.scb1 as fn2
d.text(fn2, 'hi', 100, 100)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
KeyError: WIDTH

I can put a WIDTH value in the font file, but then get: KeyError: FIRST and then KeyError: FONT after trying to manually fix this.

Here's the output file:

```

### https://fontmeme.com/fonts/scoreboard-font/
# -*- coding: utf-8 -*-
# Converted from /home/john/Downloads/scoreboard.ttf using:
#     ./font2bitmap.py /home/john/Downloads/scoreboard.ttf 16 -width 16 -s '“0123456789ABCEDFGHIJKLMNOPQRSTUVWXYZabcdefghijklmonpqrstuvwxyz”'

MAP = "“0123456789ABCEDFGHIJKLMNOPQRSTUVWXYZabcdefghijklmonpqrstuvwxyz”"
BPP = 1
HEIGHT = 11
MAX_WIDTH = 9
_WIDTHS = \
    b'\x07\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08'\
    b'\x08\x08\x08\x08\x08\x08\x08\x08\x09\x08\x08\x08\x08\x08\x08\x08'\
    b'\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08'\
    b'\x08\x08\x08\x09\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x07'

OFFSET_WIDTH = 2
_OFFSETS = \
    b'\x00\x00\x00\x4d\x00\xa5\x00\xfd\x01\x55\x01\xad\x02\x05\x02\x5d'\
    b'\x02\xb5\x03\x0d\x03\x65\x03\xbd\x04\x15\x04\x6d\x04\xc5\x05\x1d'\
    b'\x05\x75\x05\xcd\x06\x25\x06\x7d\x06\xd5\x07\x2d\x07\x85\x07\xdd'\
    b'\x08\x35\x08\x98\x08\xf0\x09\x48\x09\xa0\x09\xf8\x0a\x50\x0a\xa8'\
    b'\x0b\x00\x0b\x58\x0b\xb0\x0c\x08\x0c\x60\x0c\xb8\x0d\x10\x0d\x68'\
    b'\x0d\xc0\x0e\x18\x0e\x70\x0e\xc8\x0f\x20\x0f\x78\x0f\xd0\x10\x28'\
    b'\x10\x80\x10\xd8\x11\x30\x11\x88\x11\xeb\x12\x43\x12\x9b\x12\xf3'\
    b'\x13\x4b\x13\xa3\x13\xfb\x14\x53\x14\xab\x15\x03\x15\x5b\x15\xb3'

_BITMAPS =\
    b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x07\xff\xbe\x1e\xee\xee\xee'\
    b'\xee\xee\x5f\x3b\xff\xff\xbf\x3f\xbf\xbf\xbf\xbf\xbf\xbf\xbf\xff'\
    b'\xff\x1e\x4f\xef\xcf\x1f\x7e\xfe\x0e\x0f\xff\xff\x3e\xbf\xdf\xbc'\
    b'\x0f\xdf\xde\xdf\x3f\xf7\xff\xdf\x9f\x5f\x5e\xde\x0f\xdf\xdf\xdf'\
    b'\xff\xff\x1f\x1e\xfe\xbf\xdf\xff\xde\x1f\x3f\xff\xff\xbf\x3f\x7e'\
    b'\x9c\x0e\x6e\xee\x5f\x1f\xfb\xf6\x1e\x1f\xdf\xbf\xbf\x7f\x7f\x7f'\
    b'\x7f\xf7\xff\x3f\x1e\xdf\x3c\x0e\xde\xde\xdf\x3f\xf7\xff\x3e\x1e'\
    b'\xde\xde\x1f\x5f\xdf\x3f\x7b\xfb\xf7\x3f\x3f\x3f\xbe\xde\xde\x1e'\
    b'\xdd\xef\xf7\xfe\x3e\x9e\xde\xde\x0e\xde\xde\x1e\x3b\xfb\xf7\x3e'\
    b'\x1c\xfc\xfc\xfc\xfe\xde\x1f\xff\xf7\xf6\x1e\x1e\xfe\xfe\x3e\xfe'\
    b'\xfe\x1e\x1f\xf7\xfe\x3e\x9e\xee\xec\x0e\xee\xee\x9e\x3f\xfb\xfe'\
    b'\x1e\x1e\xfe\xfe\x3e\xfe\xfe\xfe\xff\xf7\xff\x3e\x1e\xfd\xfc\x0d'\
    b'\x8e\xde\x1f\x3b\xf7\xfe\xee\xee\xee\xec\x0e\xee\xee\xee\xeb\xff'\
    b'\xff\xff\xff\xff\x7f\x7f\xbf\xff\xff\xff\xff\xff\xdf\xdf\xdf\xdf'\
    b'\xdf\xdf\xdf\xbf\x7b\xff\xfe\xde\xde\xbe\xbe\x3e\x3e\x9e\xde\xdb'\
    b'\xff\xf6\xfe\xfe\xfe\xfe\xfe\xfe\xfe\x1e\x1f\xff\xfd\xec\xcc\xcc'\
    b'\xcc\x2d\x2d\x2d\xed\xef\xff\xfb\x75\x9a\xcd\x62\xb5\x5a\xad\x96'\
    b'\xcb\x74\xfe\xff\xe7\xc1\xdd\xbd\xbd\x9d\xd9\xc3\xff\x7f\x7e\xc7'\
    b'\xd3\xdb\xdb\xc1\xc7\xdf\xdf\xdf\xff\xfe\xe7\xcb\xdf\xdf\x81\xdf'\
    b'\xd3\xe3\xff\xff\xff\x8f\x93\xbb\xbb\x81\x87\xb7\xbb\xbb\xfe\x7f'\
    b'\xe7\xd3\xdf\xcf\xf7\xfb\xfb\xc3\xff\xfe\xfe\xc3\xc3\xf7\xf7\xf7'\
    b'\xf7\xf7\xf7\xf7\x7f\x7f\xdd\xdd\xdd\xdd\xdd\xdd\xdd\xc9\xe3\xff'\
    b'\x7f\xbd\xdb\xdb\xdb\x81\xf7\xf7\xe7\xe7\xfe\xfe\xbd\xa5\xa5\xa5'\
    b'\x81\xf7\xdb\xdb\xdb\x7f\xff\xdb\xdb\xf3\xe7\x81\xe7\xd3\xdb\xdb'\
    b'\x7e\xfe\xdb\xdb\xdb\xf7\x81\xe7\xe7\xe7\xe7\x7e\xff\xc1\xc1\xfb'\
    b'\xf3\xe7\xef\xcf\xc1\xc1\x7f\x7e\xe7\xe7\xe7\xf7\xdb\xdb\xc3\xdb'\
    b'\xbd\xfe\xff\xc7\xd3\xdb\xdb\xc1\xdb\xdb\xc3\xc7\x7f\x7e\xe7\xc3'\
    b'\x9f\x9f\x9f\x9f\xdb\xc3\xff\xfe\xff\xc7\xd3\xdd\xdd\x81\xdd\xdd'\
    b'\xd3\xc7\xff\xfe\xc3\xc3\xdf\xdf\xc7\xdf\xdf\xc3\xc3\xfe\x7f\xc3'\
    b'\xc3\xdf\xdf\xc7\xdf\xdf\xdf\xdf\xfe\xff\xe7\xc3\xdf\xbf\x81\xb1'\
    b'\xdb\xc3\xe7\x7e\xff\xdd\xdd\xdd\xdd\x81\xdd\xdd\xdd\xdd\x7f\xff'\
    b'\xff\xff\xff\xef\xef\xf7\xff\xff\xff\xff\xff\xfb\xfb\xfb\xfb\xfb'\
    b'\xfb\xfb\xf7\xef\x7f\xff\xdb\xdb\xd7\xd7\xc7\xc7\xd3\xdb\xdb\x7f'\
    b'\xfe\xdf\xdf\xdf\xdf\xdf\xdf\xdf\xc3\xc3\xff\xff\xbd\x99\x99\x99'\
    b'\x85\xa5\xa5\xbd\xbd\xff\xff\xe7\xc1\xdd\xbd\xbd\x9d\xd9\xc3\xff'\
    b'\x7f\xff\x6e\xb3\x59\xac\x56\xab\x55\xb2\xd9\x6e\x9f\xcf\xd8\xfa'\
    b'\x7b\x7b\x78\x38\xfb\xfb\xfb\xff\xff\xdc\xf9\x7b\xfb\xf0\x3b\xfa'\
    b'\x7c\x7f\xff\xff\xf1\xf2\x77\x77\x70\x30\xf6\xf7\x77\x7f\xcf\xfc'\
    b'\xfa\x7b\xf9\xfe\xff\x7f\x78\x7f\xff\xdf\xd8\x78\x7e\xfe\xfe\xfe'\
    b'\xfe\xfe\xfe\xef\xef\xfb\xbb\xbb\xbb\xbb\xbb\xbb\xb9\x3c\x7f\xef'\
    b'\xf7\xbb\x7b\x7b\x70\x3e\xfe\xfc\xfc\xff\xdf\xd7\xb4\xb4\xb4\xb0'\
    b'\x3e\xfb\x7b\x7b\x6f\xff\xfb\x7b\x7e\x7c\xf0\x3c\xfa\x7b\x7b\x6f'\
    b'\xdf\xdb\x7b\x7b\x7e\xf0\x3c\xfc\xfc\xfc\xef\xdf\xf8\x38\x3f\x7e'\
    b'\x7c\xfd\xf9\xf8\x38\x2f\xe0\x00\x00\x00\x00\x00\x00\x00\x00\x00'

WIDTHS = memoryview(_WIDTHS)
OFFSETS = memoryview(_OFFSETS)
BITMAPS = memoryview(_BITMAPS)

```

Have I done something incorrectly? Thanks in advance!

russhughes, 2023-04-10:
Use the write() method with fonts from font2bitmap. The text() method is used with the older BIOS bitmap fonts.

jouellnyc, 2023-04-14:
Right. I see now. Thank you sir!

closed 2023-04-14 by jouellnyc