russhughes/st7789_mpy #126: Font dictionary could not fully pass to C module if over roughly 128 lines

nspsck, 2023-04-21:
Am almost certain that the pointer in the C module has some limits in memory access. As only the first half of vga2 fonts can be displayed fully. 
The issue is probably caused by
`mp_obj_t font_data_buff = mp_obj_dict_get(dict, MP_OBJ_NEW_QSTR(MP_QSTR_FONT));`
in line 983. This creates a "huge" `mp_obj_t`, of which should be generally avoided. Of cause, patrially it is probably also my fault trying to use the library on an esp32 without PSRAM, you can probably get away with devices that has more RAM or PSRAM.

russhughes, 2023-04-22:
You can display the high ASCII characters (> 127) by passing an integer instead of a string to text(). You can't pass them in a string without potentially causing a UnicodeError.

```
# Draw a box in the upper left corner of the display
import vga2_16x32 as font2
tft.text(font, 0xda, 0, 0)
tft.text(font, 0xbf, font.WIDTH, 0)
tft.text(font, 0xc0, 0, font.HEIGHT)
tft.text(font, 0xd9, font.WIDTH, font.HEIGHT)
```

If the fonts are frozen into the firmware as frozen bytecode, they will use very little memory as they reside in Flash. Using them from a .py or .mpy file will use much more memory as they will reside entirely in ram.

```
Free Memory Test
Before font import: 8084320
 flash font import: 8084144 used 176
      After Text(): 8084096 used 224
          After gc: 8084112 used 208
  .py font2 import: 8067024 used 17296
          After gc: 8067520 used 16800
```

nspsck, 2023-04-22:
Thank you for the quick response! As I can confirm, this works! 

But also this raised some question.. It will be greatly appreciated if you could help me out!

Well, actually, my first thought also was the unicode causing it, since the ASCII table in C is only defined till 127. However, as I had to display a temperature, I have to send `°C` togather with some numbers. So I triedsomething like this:
`tft.text(font, '\xf8''C', font.WIDTH, font.HEIGHT)`
Appearently, micropython can not pass the '\xf8' as raw 2 bytes that represents it's integer value? How can I work around this? 
What I have tried so far is something like this:
```python
text_length = len(fill_list[2])
tft.text(font_l, 0x8f43, 18 + font_l.WIDTH * text_length, 70, 
         DAY_TEXT, ORANGE)
```
Basically I used the integer value of the text and modified 
```C
if (mp_obj_is_int(args[2])) {
        mp_int_t c = mp_obj_get_int(args[2]);
        single_char_s[0] = c & 0xff;
        str = single_char_s;
    } else {
        str = mp_obj_str_get_str(args[2]);
    }
```
this section to print the text as a whole, but am then only limited to 64bit/8bit = 8 letters... 

So, my second thought will be: implement another function to pass the text as buffer in raw bytes of which I can totally contol and ensure it is going to be "correctly coded" in C. Is this nessasary? 

Thank you very much in advance!

PS. I am already using the fonts as frozen modules and it's amazing!

russhughes, 2023-04-22:
I will look at modifying it to accept bytes as well.

ExperiMentor, 2023-04-22:
Not related to the problem with fonts here, but to be pedantic, when showing a temperature in centigrade there should **NOT** be a degree symbol. Correct notation is 23C or 23 C.

nspsck, 2023-04-23:
Thank you very much!

closed 2023-04-23 by nspsck