russhughes/st7789_mpy #131: LiLyGO T-Display S3 support?

flashypepo, 2023-06-01:
Hi @russhughes,

Very nice repo! Recommended from several persons.

I'll try to get the display working on Lilygo T-Display-S3 in micropython. I use a pre-compiled Micropython v1.20 from folder GENERIC_S3_SPIRAM_OCT (Micropython v1.20.0).

I have adopted a tft_config.py to the pins of T-Display-S3, according to [LilyGO's github repo](https://github.com/Xinyuan-LilyGO/T-Display-S3)

I only get a blank display, nothing happens, neither backlight_on nor backlight_off do anything. I've tried (hardware)SPI and SoftSPI from module machine. Sample tft_config code see below.

Just a minute ago, I've found (Circuitpython repo) the display is driven by a parallell-bus (I8080?), so my effort to adopt tft_config thru SPI might be useless.

My first issue is: Do you have some clues how to proceed?
However, maybe my issue is: could you support T-Display S3 protocol? Implementing a parallel communication protocol is beyond my knowledge. I would certainly test any setup.

Kind regards,
Peter
(from the Netherland)

Code experiment:
```
# helper
def getSPI(id=2, baudrate=30_000_000, mosiPin=19, sckPin=18, misoPin=16):
    # Hardware SPI
    spi = SPI(id,
              baudrate=baudrate,
              polarity=1, phase=1,
              mosi=Pin(mosiPin), sck=Pin(sckPin), miso=Pin(misoPin)
              )
    return spi

def config(rotation=0, buffer_size=0, options=0):
    spi = getSPI(id=1, mosiPin=13, sckPin=12, misoPin=11)
    print(spi) # debugging
    Pin(15, Pin.OUT, value=1)  # LCD-Power_On
    return st7789.ST7789(
        spi,
        170, 320,
        reset=Pin(5, Pin.OUT), cs=Pin(6, Pin.OUT), dc=Pin(7, Pin.OUT),
        backlight=Pin(38, Pin.OUT),
        rotation=rotation,
        options=options, buffer_size= buffer_size)

    tft = config(1)  # PP: landscape
    # show one image
    image = 'nasa01.jpg'
    image_file = "clock_{}x{}/{}".format(tft.width(), tft.height(), image)
    print(image_file)  # debugging
    tft.jpg(image_file, 0, 0, st7789.SLOW)
```

russhughes, 2023-06-01:
I would recommend the https://github.com/russhughes/s3lcd driver for that board.

flashypepo, 2023-06-02:
Hi @russhughes,

Aha, that is much, much better. I've flashed the proper firmware (t-display-s3) and some examples (hello, nasa-clock). It works flawless and it looks very nice!! Thank you!

Kind regards, Peter

russhughes, 2023-06-02:
Good to hear.

closed 2023-06-02 by russhughes