russhughes/st7789_mpy #143: Compile error for missing extmod/machine_spi.h when micropython 1.22 will be released

ricksorensen, 2023-10-26:
In `st7789.c`, the file `<micropython>/extmod/machine_spi.h` is included.  In recent (20231025) commits to `micropython` for cleaning up the code, this file was removed and  `modmachine.h` now includes those definitions.  This change impacts:

- `st7789.c`
- `png/pngle.c`

I have now compiled successfully (using SEEED WIO Terminal) - but still need to test with actual display.  Not sure about side effects either.

russhughes, 2023-10-27:
Thanks for the information, I hope to update my build system this weekend.

vintxo, 2023-11-03:
same problem here, hope you will fix this soon! thank you

ricksorensen, 2023-11-03:
Not a simple fix for a a system that wishes to support multiple instances of micropython!
I see in <micropython>/py/mpconfig.h there are the following defines fro branch (version)1.21:
```C++
// Current version of MicroPython
#define MICROPY_VERSION_MAJOR 1
#define MICROPY_VERSION_MINOR 21
#define MICROPY_VERSION_MICRO 0
```

and for the current (20231103) master branch:
```c++
// Current version of MicroPython. This is used by sys.implementation.version
// as well as a fallback to generate MICROPY_GIT_TAG if the git repo or tags
// are unavailable.
#define MICROPY_VERSION_MAJOR 1
#define MICROPY_VERSION_MINOR 22
#define MICROPY_VERSION_MICRO 0
#define MICROPY_VERSION_PRERELEASE 1
```
I suppose a conditional compile could be used to determine which file to include, for example:
```c++
....
// need this to get definitions I believe
#include "mp/mpconfig.h" 
...   
#if MICROPY_VERSION_MAJOR == 1 && MICROPY_VERSION_MINOR > 21 
#include "extmod/modmachine.h"
#else
#include "extmod/machine_spi.h"
#endif
```

commit 40d6990d (2023-11-05): Fix #143 use modmachine.h for MicroPython > 1.21
closed 2023-11-05 by russhughes