russhughes/st7789_mpy #144: I can not found the st7789.py file

omrfrkmll, 2023-11-01:
Greetings everyone,

I am new to micropython and I am having trouble understanding how to work with the st7789 library. I have started working with micropython and found the "something_for_micropython.py" file which I uploaded to my device and imported into the "main.py" file. However, for this particular project, I am unsure of how to proceed. Is there any "st7789.py" file or should I directly flash the "exp_device.uf2" to my board? Any help would be greatly appreciated. Thank you.

russhughes, 2023-11-05:
This driver has to be compiled with the MicroPython firmware and then flashed to your device. Several precompiled firmware files are available in the firmware folder.

omrfrkmll, 2023-11-06:
Thanks for your reply.

closed 2023-11-06 by omrfrkmll