russhughes/st7789_mpy #145: OSError: [Errno 2] ENOENT trying some examples

hajulied, 2023-11-07:
I have an ESP32 WROOM with a display 1,69 ST7789. 
Installed is   MicroPython v1.20.0-dirty on 2023-06-10; ESP32 GENERIC-7789 with ESP32

I tried some of the examples. "tiny_hello.py" and "chango.py" work fine.

"clock,py" gives the follwing error:

Traceback (most recent call last):
  File "<stdin>", line 191, in <module>
  File "<stdin>", line 118, in main
OSError: [Errno 2] ENOENT

line 118:     "tft.jpg(image_file, 0, 0, st7789.SLOW)

"watch.py" is similar:
  File "<stdin>", line 147, in <module>
  File "<stdin>", line 47, in main
OSError: [Errno 2] ENOENT

line 47:     tft.jpg(face, 0, 0, st7789.SLOW)
   
what can I do?

russhughes, 2023-11-07:
It should be better documented. The ENOENT error means the program is looking for a file it can't find. The clock program reads images from a clock directory for the backgrounds. The directory is named clock_{width}x{height} where {width} and {height} are the width and height of the display in landscape mode.   For example, a 320x240 display would read from a directory named clock_320x240.  The repo has sample directories for 128x128, 160x128, 240x135, 240x240, and 320x240 displays.  Create a directory for your screen size and copy the images into that directory.  You will also need the pacifico40.py in the same directory as the clock.py file.

The watch.py program needs a face_{width}x{height}.jpg image in the same directory as the watch.py program. The repo has several sample face images you can use. The create_face_jpg.py file can create a custom-sized face image using a PC.

hajulied, 2023-11-09:
Thank you, it works fine now for the above mentioned resolutions. 
Is it possible to extend the driver to a resolution of 240x280?
My display (240x280) works with 240x240 and 320x240 but it does not fit.

russhughes, 2023-11-09:
To add a custom-sized display, create an orientation table and pass it to the ST7789 method as the named parameter `rotations`. See the [tft_config.py](examples/configs/adafruit_320x172/tft_config.py) file in the examples/configs/adafruit_320x172/ directory for an example of a 320x172 display. 

The orientation table lists tuples for each rotation used to set the MADCTL register, display width, display height, start_x, and start_y values.  I can help you figure out the necessary values if needed.  Can you give me a link to the display you are using?

hajulied, 2023-11-14:
I was a few day on tour. 
Thank  you for your tip. 
I bought the display in Germany at AZ...
https://cdn.shopify.com/s/files/1/1509/1638/files/1_69inchTFT_datasheet.pdf?v=1694076008
I modified the adafruit example with "240, 280, 0,0    280, 240, 0,,0) and changed the resolution in SPI(... 240, 280)
But then there were some unused rows at the bottom.
Then I changed the values  from 280 to 300 , and now it works fine an the whole screen is used.
``
custom_rotations = [
    (0x00, 240, 300, 0, 0),
    (0x60, 300, 240, 0, 0),
    (0xc0, 240, 300, 0, 0),
    (0xa0, 300, 240, 0, 0),
]
def config(rotation=0, buffer_size=0, options=0):
    return st7789.ST7789(
	SPI(2, baudrate=40000000, sck=Pin(18), mosi=Pin(19), miso=None),        
	    240,
        300,
[image]
[image]
[image]
[image]