russhughes/st7789_mpy #146: Failure to compile for Teensy40

robhamerling, 2023-11-09:
When building firmware with st7789_mpy for Teensy 4.0 I get the report:

<pre><b>/home/rob/upy/lib/st7789_mpy/st7789/st7789.c:</b> In function &apos;<b>write_cmd</b>&apos;:
<b>/home/rob/upy/lib/st7789_mpy/st7789/st7789.c:74:18:</b> <span style="color:#EF2929"><b>error: </b></span>comparison between pointer and integer [<span style="color:#EF2929"><b>-Werror</b></span>]
   74 |     if (self-&gt;cs <span style="color:#EF2929"><b>!=</b></span> GPIO_NUM_NC) {     \
      |                  <span style="color:#EF2929"><b>^~</b></span>
</pre>

and many similar lines.

I see in the comments in st7789.c that I may need to change type or value of GPIO_NUM_NC, and for the Teensy it probably needs to be a pointer type, perhaps &(-1) ? 
Maybe an '#ifdef" need to be added for the Teensy?

russhughes, 2023-11-09:
The GPIO_NUM_NC definition needs to be refactored. A quick workaround is to add CFLAGS_EXTRA="-DGPIO_NUM_NC=NULL" to your make command.

robhamerling, 2023-11-09:
With this suggested workaround  compilation succeeded and firmware works, thanks.