russhughes/st7789_mpy #147: Will not compile with module

RaspberryPiDude314, 2023-11-21:
Hi all, fairly new to building Micropython from source.
I've tried every variation of `make -C ports/rp2 BOARD=RPI_PICO_W USER_C_MODULES=/home/pi/pico/st7789_mpy/micropython.cmake` I can think of but no matter what, it does not compile with the library. While it does build, the library just isn't present. It seems to just ignore it whatever I do.
Help is much appreciated!

russhughes, 2023-11-22:
What version of MicroPython are you trying to compile?

russhughes, 2023-11-22:
I used:

```
make -C ports/rp2 BOARD=RPI_PICO_W USER_C_MODULES=/Volumes/MicroPython/st7789c/micropython.cmake
```

And it was successful. You should be able to find the following lines in the make output:
```
Including User C Module(s) from /Volumes/MicroPython/st7789c/micropython.cmake
Found User C Module(s): usermod_st7789
```
If they don't appear, the module source is not being found.

RaspberryPiDude314, 2023-11-22:
I did eventually locate the problem- due to some bug with the compiler, it will not re-check for external c modules since the last build in that directory. After figuring this out, all O had to do was run make clean and make it again and it worked as intended. There's unfortunately not much information about this bug but that was the problem.

closed 2023-11-22 by RaspberryPiDude314