russhughes/st7789_mpy #152: rp pico no display output

omrfrkmll, 2024-01-10:
Hello;
I am currently using rp pico and bought this st7789 lcd:
[image]

I still trying to get any pixel light on except backlight. while trying https://github.com/russhughes/st7789py_mpy this library, the screen is working but on compiled firmware does not work. 
here is my configs while using https://github.com/russhughes/st7789py_mpy library:
```
import st7789py as st7789

display = st7789.ST7789(
        SPI(1, baudrate=62500000, sck=Pin(10), mosi=Pin(11)),
        240,
        240,
        reset=Pin(14, Pin.OUT),
        cs=Pin(13, Pin.OUT),
        dc=Pin(12, Pin.OUT),
        backlight=Pin(15, Pin.OUT),rotation=2)
```
and here is the compiled uf2 file uploaded firmware configs:
```
import st7789

def scrconfig(rotation=1, buffer_size=0, options=0):
    return st7789.ST7789(
        SPI(1, baudrate=62500000, sck=Pin(10), mosi=Pin(11)),
        240,
        240,
        reset=Pin(14, Pin.OUT),
        cs=Pin(13, Pin.OUT),
        dc=Pin(12, Pin.OUT),
        backlight=Pin(15, Pin.OUT),
        rotation=rotation,
        options=options,
        buffer_size=buffer_size)
tft = scrconfig(1)
```
I cannot find where did I go wrong. Could someone help please...

russhughes, 2024-01-10:
Make sure that you are calling tft.init() to initialize the display. The pure Python driver does this automatically, and the C version does not.

omrfrkmll, 2024-01-10:
Yes, sir. I have included this command in my code. However, I forgot to add the "tft.init()" line here. Even after trying to delete or double it, the code still does not work. I am confused about what is included in the https://github.com/russhughes/st7789py_mpy library and what is not contained in the compiled version.

flashypepo, 2024-02-26:
@omrfrmll: Maybe you have already solved it. Maybe next will help you. I do have a ST7789 TFT-display working on Pico using the great russhughes st7789_mpy driver (built-in custom firmware), but using SPI0 (on a Grove PICO shield). 

You seems to use SPI1, but I've two remarks: SPI1_CS=Pin(9)  and I've read that baudrate is at most 30_000_000. Pins according to Raspberry PICO pin layout at https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html

PS. I'm a software person, not hardware. It might be acceptable using other pins then the hardware-SPI pins.

Kind regards, Peter