russhughes/st7789_mpy #163: Adding network capabilities.

Asleep123, 2024-05-14:
Hi.

The prebuilt firmware images provided use an outdated version of MicroPython and do not have the built in networking libraries. I was wondering if I could just add ST7789 onto an already existing updated MicroPython firmware instead of building my own, because that seems sort of intricate.

gitcnd, 2024-07-31:
Ditto that.

It's possible to build native "C" standalone .mpy modules (example - the "factorial" demo)... that would definitely be worth exploring so this driver can ship as a file, instead of an entire firmware...

If you're still stuck on this, let me know - I've got it building fine, and can send you a binary if you want.

asherevan, 2024-08-27:
If you could also send it to me, that would be great!