russhughes/st7789_mpy #166: Text problem

asherevan, 2024-07-27:
When you try to draw text to the very top of the screen, it only shows random pixels there instead.

[image]

sriram-rs, 2024-08-27:
Can you add a picture of just the screen without any contents (text/shapes) on it?

Is this how it is shown always or you see this pixels only when you try to draw?

asherevan, 2024-08-27:
[image]

It is only shown when I try to draw text (Text does usually work, it is only sometimes)

sriram-rs, 2024-08-29:
From my little observations, I had also come across this and for me it was due to wrong height and width values that were provided on the tft_config.py. I suggest that you check these values once. 

The other possibility is that this could be due to (font.py) bitmap file that you use. You may need to check font's height and width.

Hope this helps!!