russhughes/st7789_mpy #168: Update the firmwares please

asherevan, 2024-08-27:
Could you update the firmwares please?

I am unable to compile them myself right now, so it would be great if you would!

PIBSAS, 2024-09-10:
@asherevan 

### Install dependencies:

```bash
sudo apt-get update && sudo apt-get install -y gcc-arm-none-eabi libssl-dev python3 python3-pip python3-venv build-essential libffi-dev git pkg-config
```

### Clone the nice work:

```bash
git clone https://github.com/russhughes/st7789_mpy.git
```

### Clone MicroPython:

```bash
git clone --recurse-submodules https://github.com/micropython/micropython.git
```

Build MicroPython+ST7789:
```bash
cd micropython
make -C mpy-cross
cd ports/rp2
make USER_C_MODULES=../../../st7789_mpy/st7789/micropython.cmake
make submodules
make
```

Note: If make user line fail, change ``../../../`` for the full path, example: ``/home/your_user/st7789_mpy/st7789/micropython.cmake``.

#### The results will be in:
``micropython/ports/rp2/build-RPI_PICO/firmare.uf2``

And Finally Star this repo to thanks russhughes.

asherevan, 2024-09-18:
If I compile it for esp32, what would the file be called?

PIBSAS, 2024-09-18:
oops sorry, i was compiling for that board and forget the rest of the world, the steps for the esp32 are in the readme. but i will try and let those steps if the compile works. excuse for the wrong answer

asherevan, 2024-09-18:
The compiling worked, but I’m just wondering which of the files is the
actual firmware.

PIBSAS, 2024-09-18:
Are the three micropython.bin, bootloader.bin and partition-table.bin. and you flash like the instructions in micropython site says. if is a [ESP32](https://micropython.org/download/ESP32_GENERIC/), if is [ESP32-S2](https://micropython.org/download/ESP32_GENERIC_S2/) if is a [ESP32-S3](https://micropython.org/download/ESP32_GENERIC_S3/), if is a [ESP32-C3](https://micropython.org/download/ESP32_GENERIC_C3/) you dont specify which one he!. You need to have Python installed, a [Python Virtual Environment](https://docs.python.org/3/library/venv.html), and install esptool (pip install esptool).

russhughes, 2024-09-18:
> If I compile it for esp32, what would the file be called?

firmware.bin in the ports/esp32/build-{your_board_name}

asherevan, 2024-09-18:
> > If I compile it for esp32, what would the file be called?
> 
> firmware.bin in the ports/esp32/build-{your_board_name}

Thanks! I will submit a pull request with it.

closed 2024-10-13 by asherevan