russhughes/st7789_mpy #174: Cannot build firmware: USER_C_MODULES does not exist.

asherevan, 2024-12-06:
When I try to follow your instructions to compile the firmware on ubuntu, it works until I try to run:

`
make USER_C_MODULES=../../../../st7789_mpy/st7789/micropython.cmake
`

the build fails and returns this error:

```
FAILED: /home/myuser/micropython/mpy-cross/build/mpy-cross
cd /home/myuser/micropython/ports/esp32/build-ESP32_GENERIC/esp-idf/main_esp32 && make -C /home/myuser/micropython/mpy-cross
make[1]: Entering directory '/home/myuser/micropython/mpy-cross'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
../py/py.mk:37: *** USER_C_MODULES doesn't exist: /home/myuser/st7789_mpy/st7789/micropython.cmake.  Stop.
```

How can I make it compile?

dobodu, 2024-12-23:
Hi,

I guess you are using ESP-IDF ? can you tell u which version ?

The problem is, I think, not related to st7789 library but more to ESP_IDF that seems to miss something in the directory path configuration.  Maybe trying an older ESP-IDF version should correct the problem.

asherevan, 2024-12-23:
I am using ESP-IDF version 5.2.2

dobodu, 2024-12-23:
Well, obvisously I got the same issue with 5.2.3 (for st7789 library, but also for my lilygo_amoled library...)
Maybe 5.1 would be right even if they already did things on the C/C++ compiler.
I'll try to find the right correction to do since I don't like being stuck in older version... Let's keep in touch.

dobodu, 2024-12-23:
[https://github.com/micropython/micropython/issues/12972](https://github.com/micropython/micropython/issues/12972)

dobodu, 2024-12-23:
would you try 

`make USER_C_MODULES=../../../../st7789_mpy/st7789 `

asherevan, 2024-12-23:
It worked! Thanks!

closed 2024-12-23 by asherevan