russhughes/st7789_mpy #181: Who can teach me how to get ST7789.py step by step?

h3clikejava, 2025-04-13:
I'm a beginner and I have no idea how to start. Can anyone teach me? Thank you very much!

bcw222, 2025-07-26:
For better performance, this driver is shipped as a C extension; consequently, there is no file named `ST7789.py`.  
You must:

1. Select the correct, port-specific pre-built firmware (or build it on your own if necessary).  
2. Flash that firmware onto your microcontroller.

Once the firmware is in place, the driver can be used directly in MicroPython via:

```python
import st7789
```

asherevan, 2025-07-26:
There is no way to get a st7789.py file (that I know of) but you can look
in the README.md file for a list of the attributes (with needed arguments)
of the library.

> Message ID: ***@***.***>
>